ciniml/WireGuard-ESP32-Arduino #2: [Question] Compile filed "xtensa-esp32-elf-as: unrecognized option '-x'"

mrkeuz, 2021-10-02:
Hi, first of all, thanks for your effort! 🚀 

I have a question. We have idea (early WIP) tried to integrate and build project (`ESPHome`) with your library, but unfortunately, compile filed with `xtensa-esp32-elf-as: unrecognized option '-x'` error.
PlatformIO upgraded and latest version (5.2.0)

Could you help to find a way to fix this issue?
Thanks in advance.

My config (I tried on fresh project):
```ini
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
lib_deps = ciniml/WireGuard-ESP32 @ ~0.1.2
```

Here output of error
```console
➜   export LC_ALL=C    
➜   platformio run -t upload          
Processing esp-wrover-kit (platform: espressif32; board: esp-wrover-kit; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (3.3.1) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WireGuard-ESP32> 0.1.2
Building in release mode
Compiling .pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/cortex_m0_mpy121666.s.o
Compiling .pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/cortex_m0_reduce25519.s.o
xtensa-esp32-elf-as: unrecognized option '-x'
Compiling .pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/mul.s.o
xtensa-esp32-elf-as: unrecognized option '-x'
xtensa-esp32-elf-as: unrecognized option '-x'
*** [.pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/cortex_m0_mpy121666.s.o] Error 1
Compiling .pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/sqr.s.o
*** [.pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/mul.s.o] Error 1
*** [.pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/cortex_m0_reduce25519.s.o] Error 1
xtensa-esp32-elf-as: unrecognized option '-x'
*** [.pio/build/esp-wrover-kit/libca3/WireGuard-ESP32/crypto/cortex/sqr.s.o] Error 1
============================================================================== [FAILED] Took 0.83 seconds ==============================================================================

```

mrkeuz, 2021-10-02:
Little update, I found how remove wrong flags. But it does not help.
- Logs here: 
  https://github.com/esphome/feature-requests/issues/1444

mrkeuz, 2021-10-03:
Found solution. 

In the end reveals that `src/crypto/cortex` is some platform-specific implementation of cryptography and even not used (call of asm implementation `crypto_scalarmult_curve25519` is commented):

https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/ce65891c207141f9c75dd318a6faa53c6c20444e/src/crypto.h#L20-L21

It can exclude from library safety. I don't find easy way to exclude some lib files from build via platformio.ini flags, it can achieve little tricky via python script (extra_scipts = ... flag). But I decided just made fork and fix library.

I can create PR for this. As you support only `esp32` right now, so files can be deleted. 

In the future, for such cross-platform optimizations it can be done via `extra_scrips`, I think.
(see: https://docs.platformio.org/en/latest/librarymanager/config.html)

ciniml, 2021-10-03:
Thanks for investigating the problem.

This is an Arduino library for ESP32, so ESP32-specific resolution is OK.

Could you post a PR for the changes? I will check and merge it.

ciniml, 2021-10-03:
Thanks for posting PR.

I've confirmed that the PR solves this issue. Thanks!

closed 2021-10-03 by ciniml

mrkeuz, 2021-10-03:
Glad to help. Love `WireGuard`. Nice project!
I wish you successful development and fewer bugs! 🚀