ciniml/WireGuard-ESP32-Arduino #9: MQTT via WireGuard

PricelessToolkit, 2021-11-07:
Hi, thanks for WireGuard Implementation on esp32.
I wanted to ask, for to not suffer for several days with debugging MQTT via WireGuard is working?
did not find information anywhere.
Thanks

ulikoehler, 2021-12-31:
Plain old HTTP via Wireguard works, see my example here:
https://techoverflow.net/2021/12/31/esp32-wireguard-example-with-http-access-over-wireguard-platformio/

So you can be sure that MQTT works as well. Feel free to use my example as a starting point.

uGiFarukh, 2023-04-22:
@ulikoehler looks like none of the http or mqtt traffic is passing through the wireguard tunnel. I followed your guide but it seems to be not working. Maybe something to do with traffic forwarding needs to be somehow enabled in esp32 LwIP stack for this to work?