ciniml/WireGuard-ESP32-Arduino #10: DNS & disconnect

hydrafi, 2021-11-16:
Hi, thanks for WireGuard Implementation on esp32.
I am trying and it runs well. 
I noticed that all traffic goes through the VPN. It would be helpful if at least DNS requests used the local STA connection. 
Also a "disconnect" function would be useful.
Thanks

MaxT

emonindonesia, 2021-12-03:
I second to be actually set allowed IP addresses. There is allowed_ip in the code, but I don't see a clean way to set that.
Wireguard configs usually have something like
`AllowedIPs = 10.0.1.0/24`

ciniml, 2021-12-06:
According to the lwIP implementation, traffics are routed based on the interface netmask by default.
So I think DNS traffics go through the underlying interface if the DNS server is at the same network of the underlying interface.

Forwarding via a gateway in the non-default interface is not supported by lwIP, since the lwIP does not have any routing table.
The routing of the lwIP is done based on the netmask of the interfaces. 

I think a "disconnection" could be implemented by restoring the default interface to the default one and shutdown the Wire Guard interface.

https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/main/src/WireGuard.cpp#L100

To restore the default interface before Wire Guard connection, we must save the default interface before updating it by calling `netif_set_default`.
It seems to be done by just storing `netif_default` global variable which lwIP exposes, but I haven't tried it yet.

ciniml, 2021-12-08:
I've implemented `WireGuard::end()` function to shutdown the WireGuard interface.

https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/10-disconnect/src/WireGuard.cpp#L122

There is an example of the `end()` function.

https://github.com/ciniml/WireGuard-ESP32-Arduino/blob/10-disconnect/examples/disconnect/disconnect.ino

ciniml, 2021-12-11:
I've released WireGuard for ESP32 Arduino Library 0.1.5. 

This version supports shutting down the WireGuard interface by calling `end()` function.