ciniml/WireGuard-ESP32-Arduino #12: How to generate device public key

rjjrbatarao, 2021-12-11:
Is there any way of getting the device public key? theres only peer public key available but I dont know where to get the device public key

ciniml, 2021-12-11:
This library does not have any key generation feature, so you must generate a device private and public key pair somewhere else. 
e.g. Generating the device private and public keys on a PC with [this instruction](https://www.wireguard.com/quickstart/).

Then you can embed the generated private key to the sketch and register the generated public key to the WireGuard remote peer.

rjjrbatarao, 2021-12-11:
Thank you I'll follow the instruction right away.

rjjrbatarao, 2021-12-11:
I found an online tool t generate private nad public key https://www.wireguardconfig.com/  hope this helps.

closed 2021-12-11 by rjjrbatarao