ciniml/WireGuard-ESP32-Arduino #18: Unexpected reboot after calling wg.end()

kerokerowood, 2022-03-17:
Hi, I'm enjoying WireGuard-ESP32@^0.1.5 and thanks for this great library for ESP32.  I'm using this library in the following environment:
- VSCode + PlatformIO (latest) with Arduino framework
- Issue occurs with any of M5Stack / M5Core2 / M5AtomMatrix

What I am doing is 1) start everything for WireGuard communication, 2) do something thru the tunnel, and 3) stop everything in every cycle in the loop. 

```
        #include <WireGuard-ESP32.h>

        WireGuard wg;
        while(1) {
                Wifi.begin(ssid, pass);
                configTime( parameters );
                wg.begin(  wireguard parameters  );
                  (wait for about 1 second)
                WiFiClient client;  
                     do something thru WireGuard tunnel
                client.stop();
                wg.end();
                WiFi.disconnect();
                vTaskDelay(waitSecond);
        }
```

The issue is **unexpected reboot** in void WireGuard::end().  In some cases it happened at every call but in other case at not every call. I could not find any dependencies for these differences.  Temporary fix I found in the end is to add **net_set_down()** just before the netif_remove().  Currently this removes the issue for all the cases I had. 

```
        void WireGuard::end() {
                    :
                netif_set_down(wg_netif);    // <--- temporary fix
                netif_remove(wg_netif);
                wg_netif = nullptr; 

                this->_is_initialized = false;
        }
```

I'm not sure if it is a correct way or not. So, I would be grateful if you could kindly give me advice about this issue.

ciniml, 2022-04-16:
Hi. 
Sorry for late response. Thanks for reporting.

Maybe your fix is correct and I will merge this fix after I check why calling `netif_set_down` is required by digging the lwIP code.

kerokerowood, 2022-04-16:
@ciniml san,  no problem. Currently I'm using local fix and don't have any critical issues.  Thank you for checking my comment and I am very grad to help you out.