ciniml/WireGuard-ESP32-Arduino #19: Forward between interfaces not working on ESP32

CatalinPuscoci, 2022-05-29:
Hello, I have a setup where I have two IPs in allowedIPs on the ESP's peer like this
`AllowedIPS = 192.168.1.0/24,192.168.69.14/32` 

The network topology is as follows

Peer B(raspi)(address in 192.168.0.0/24 and 192.168.69.0/24) <-> Peer A(ESP) (192.168.1.0/24 and 192.168.69.0/24).

With another device that is not an ESP, once net.ipv4.ip_forward=1 is set, and routes are set(the wireguard exposing the endpoint generates the route,and another static route is configured on the routers) Peer B can ping any device in Peer A's network, and viceversa. But Even after defining IP_FORWARD to 1 (technically forwarding exposed from lwip's side) I cannot seem to reach other devices in the ESP's network, with my packets being dropped.

zacchaeus80, 2022-08-02:
Hi CatalinPuscoci,

I'm grappling with what I think is the same problem.  Did you find a fix?

Can you also please explain how you set IP_FORWARD to 1 on the ESP?

Thanks.

uGiFarukh, 2023-02-06:
Having a similar issue. Trying to forward traffic through the ESP just like a VPN. I have 0.0.0.0/0 in allowed ips list for my windows wireguard client. And I have the windows wireguard client's ip address inside ESP's allowed ip list like 10.100.100.5/32. Trying to relay all traffic from PC to the internet through the ESP32.

Any possible method for doing this?

uGiFarukh, 2023-04-21:
@ciniml please help on this.