ciniml/WireGuard-ESP32-Arduino #23: PSK

psxde, 2022-09-29:
Is there an option to include a Preshared Key to the ESP Wireguard Tunnel?

tkndrm, 2022-10-01:
hi, i guess shouldn't be that difficult to implement. 
Seems like there just no way to define it with begin().

https://github.com/ciniml/WireGuard-ESP32-Arduino/search?q=preshared

zjonesz, 2022-11-25:
Did you @psxde  manage to get it working with PSK?

brummbaer, 2022-11-28:
Hi, I am now passing the preSharedKey to the "begin" and have adjusted the code to do so. However, I can't get it to work in general. I am using autoConnect. Has anyone ever gotten this to work with ESP32? Could use some help. Thanks

kienvu58, 2022-12-03:
> > Hi, I am now passing the preSharedKey to the "begin" and have adjusted the code to do so.
> 
> nice, can you pls share the code and adjustments?

I've adjusted the code to allow to specify a preshared key. Check out my fork: https://github.com/kienvu58/WireGuard-ESP32-Arduino.git

absemetov, 2023-01-23:
@kienvu58 Thanks code work!