ciniml/WireGuard-ESP32-Arduino #34: Issues when using library with GPRS/GSM modem

sveinbjorng, 2023-09-04:
Hello, I have a [LilyGo T-A7670X](https://github.com/Xinyuan-LilyGO/T-A7670X) GNSS/GSM module. The module crashes after calling wg.begin(). I'm assuming this is due to the wrong interface being selected.

trace:

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ed10d  PS      : 0x00060830  A0      : 0x800ed378  A1      : 0x3ffb94e0  
A2      : 0x3ffb21e0  A3      : 0x3ffb2768  A4      : 0x3ffb2314  A5      : 0x00000000  
A6      : 0x0000002e  A7      : 0x00000004  A8      : 0x00000000  A9      : 0x3ffb276b  
A10     : 0x3ffb2768  A11     : 0x40085ad4  A12     : 0x3ffb95cc  A13     : 0x0000003f  
A14     : 0x0000eba0  A15     : 0x0000002d  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x400881bc  LEND    : 0x400881d8  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400ed10d:0x3ffb94e0 0x400ed375:0x3ffb9520 0x400d6b36:0x3ffb9540 0x400ea559:0x3ffb9610 0x400ea760:0x3ffb9630 0x400f419c:0x3ffb9650 0x4008ac0e:0x3ffb9680

Rebooting...`

jichef, 2024-03-21:
Have you got it solved?

jichef, 2024-04-17:
> Did you found solution? Im also trying to connect with TSIM7000, but without luck.

Hi! Sorry… No. I didn't get it. When I use the GSM connection, it compiles correctly, but it does not establish a connection; however, using the same board (tsim7000) but through wifi, it connects perfectly. Therefore, it must be some problem of the library or something like that, related to the interface. I haven't been able to get a deep log to see where it gets stuck. My knowledge escapes...

sveinbjorng, 2024-04-18:
@jichef @bkobylski4 I never got this to work. It's been a while since I looked at this but I believe the core issue may be with how TinyGPSPlus provides the interface. If I remember correctly, It creates some sort of a virtual interface that this library doesn't expect.

jichef, 2024-06-12:
> @jichef @bkobylski4 I never got this to work. It's been a while since I looked at this but I believe the core issue may be with how TinyGPSPlus provides the interface. If I remember correctly, It creates some sort of a virtual interface that this library doesn't expect.

Hi! Actually I've found some light, but I don't know how to integrate this solution in Arduino IDE
[``](https://github.com/trombik/esp_wireguard/issues/39#issuecomment-2153254489)