ciniml/WireGuard-ESP32-Arduino #36: Issue with VPN connection on ESP32 using WireGuard library

mauro-midolo, 2023-12-19:
I am encountering difficulties in establishing a VPN connection using the WireGuard-ESP32 library on an ESP32 device. I have included my sample code below, but it seems not to be functioning as expected.

```
#include "WiFi.h"
#include <WireGuard-ESP32.h>
#include <ESPping.h>

static WireGuard wg;
IPAddress targetIP(10, 210, 144, 3);  // Destination IP address

char private_key[] = "<PRIVATE_KEY>";  // [Interface] PrivateKey
IPAddress local_ip(10, 210, 144, 9);            // [Interface] Address
char public_key[] = "<PUBLIC_KEY>";     // [Peer] PublicKey
char endpoint_address[] = "vpn.myhost.com";    // [Peer] Endpoint
int endpoint_port = 51820;              // [Peer] Endpoint

void setup() {
  // ... (WiFi initialization code)

  Serial.println("\nConnected to WiFi network");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Adjusting system time...");
  configTime(9 * 60 * 60, 0, "ntp.jst.mfeed.ad.jp", "ntp.nict.jp", "time.google.com");
  Serial.println("\Connecting to VPN network");

  // Initializing WireGuard connection
  wg.begin(
    local_ip,
    private_key,
    endpoint_address,
    public_key,
    endpoint_port);
  Serial.println("\nConnected to VPN network");
}

void loop() {
  // ... (Loop code)

  bool pingResult = Ping.ping(targetIP);

  if (pingResult) {
    Serial.print("Response from ");
    Serial.print(targetIP);
    Serial.print(" in ");
    Serial.print(Ping.averageTime());
    Serial.println(" ms");
  } else {
    Serial.println("Ping failed");
  }

  // ... (Other commands in the loop)
}
```

I have followed the WiFi connection process and WireGuard initialization as per the documentation, but it seems that the VPN connection fails to establish correctly. I have also included code for pinging a destination IP address.

Thank you for your assistance in resolving this issue.

wawanUnic, 2024-01-15:
Привет. Работает хорошо.
https://github.com/wawanUnic/ESP32-WireGuard

Но медленно:
Success/Wrong pings: 141/5. Time begin ping: 00:13:43
Response from 192.168.42.1 in 39.38 ms
Success/Wrong pings: 142/5. Time begin ping: 00:13:47
Response from 192.168.42.1 in 11.68 ms
Success/Wrong pings: 143/5. Time begin ping: 00:13:51
Response from 192.168.42.1 in 9.10 ms
Success/Wrong pings: 144/5. Time begin ping: 00:13:55
Response from 192.168.42.1 in 25.28 ms
Success/Wrong pings: 145/5. Time begin ping: 00:13:59
Response from 192.168.42.1 in 529.07 ms
Success/Wrong pings: 146/5. Time begin ping: 00:14:06
Response from 192.168.42.1 in 289.63 ms
Success/Wrong pings: 147/5. Time begin ping: 00:14:13
Response from 192.168.42.1 in 615.04 ms
Success/Wrong pings: 148/5. Time begin ping: 00:14:21
Response from 192.168.42.1 in 246.30 ms
Success/Wrong pings: 149/5. Time begin ping: 00:14:27

При том, что в цикле loop ничего нет, кроме пинга:

`void loop()
{
  bool pingResult = Ping.ping(targetIP);

  if (pingResult) {
    Serial.print("Response from ");
    Serial.print(targetIP);
    Serial.print(" in ");
    Serial.print(Ping.averageTime());
    Serial.println(" ms");
  } else {
    Serial.println("Ping failed");
  }`

Из 173 удачных обменов было потеряно 6.

MrKappa, 2024-02-26:
Enable the ESP32 debug options. It's possibile that you're trying to connect to a wg server that requires a preshared key. Right now the preshared key isn't managed. Try to configure your server for not requiring preshared keys.