ciniml/WireGuard-ESP32-Arduino #37: Connect to ESPAsyncWebServer / ArduinoOTA / Services on ESP32

itsyoshio, 2024-01-18:
Hi, i'm using the Example code and Outbound connections work well.

I've successfully pinged my Router (also the Wireguard Server) and used the WiFiClient to fetch a Website on my Wireguard Network over HTTP. 

The ESP32 i use as the Wireguard Client also runs a [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) (port 80) and [ArduinoOTA](https://github.com/JAndrassy/ArduinoOTA) (port 32537).

I can't ping it, i can't access the Webserver and OTA Updates also don't work within the VPN, but do run over Ethernet or Wifi directly.

Any help would be appreciated.

Shinmera, 2024-01-19:
I'm having the same issue using the standard `esp_http_server`. Outbound is OK, inbound ends up with unreachable address.

Is there some kind of routing we have to perform to make sure the packets arrive at the correct place?

MrKappa, 2024-02-26:
> In my case yes. I'm using exactly the same configuration as for every other client in the net, and the rest work just fine.

Is your device wg ip pingable? If yes, looks like the problem could be that the service isn't listening/binding to the wg created interface. Can you try to start the tunnel and then initialize/start your services? Otherwise I'd look how those services bind to the network interfaces.

Shinmera, 2024-11-05:
> Is your device wg ip pingable? If yes, looks like the problem could be that the service isn't listening/binding to the wg created interface. Can you try to start the tunnel and then initialize/start [...]

Pinging the local IP of the ESP works fine, but if I ping the WireGuard VPN IP, it says that the destination host is unreachable.

I thought to try and ping something from the ESP first, like the WireGuard server, but that also fails, so it just seems to me like it's not establishing the connection properly at all?