ciniml/WireGuard-ESP32-Arduino #43: Request for Support with Pre-shared Key

mauro-midolo, 2024-04-19:
I'm reaching out to request the implementation of pre-shared key support. 
This addition would greatly benefit users requiring secure communication. 

Could you consider integrating this feature into the project?

Thank you for your consideration.

SpaceTeddy, 2024-05-30:
This fork has already implemented the pre-shared key feature:
https://github.com/kienvu58/WireGuard-ESP32-Arduino

LockManipulator, 2025-02-05:
@SpaceTeddy When I try to use your library with the provided code in examples, I get an error saying I need 14 arguments for the .begin instead of the 5 as provided in the example. Can you give a minimally working example of your library?

SpaceTeddy, 2025-02-06:
well, I'dnt't compile the code below, but this is more and less my setup for initialisation.

//------------------------------[ WireGuard configuration ]----------------------
#include <WireGuard-ESP32.h>
#include "wireguardif.h"

static WireGuard wg;

IPAddress                 local_ip(192,168,0,100);                          // [Interface] Address, choose your local IP
char private_key[]      = "your_private_key";   // [Interface] PrivateKey
char public_key[]       = "your_public_key";   // [Peer] PublicKey
char preshared_key[]    = "your_pre_shared_key";   // [Peer] PreSharedKey
char endpoint_address[] = "your_endpint_address";                                  // [Peer] Endpoint hostname or IP
int  endpoint_port       = 50614;                                           // [Peer] Endpoint port

setup_wg(){

       
        if(!wg.is_initialized()){
            Serial.println("Initializing WG interface...");
            if( !wg.begin(
                    local_ip,
                    private_key,
                    endpoint_address,
                    public_key,
                    endpoint_port,
                    preshared_key) ) {
                Serial.println("Failed to initialize WG interface.");
                wg.end();
            }
        }
        else {
            Serial.println("Shutting down WG interface...");
            wg.end();
        }
        
        delay(500);
}