ciniml/WireGuard-ESP32-Arduino #45: ESP32 with W5500 and platform 3 compille error

proasnet, 2024-07-04:
ESP32 with connected W5500 ethernet adapter, esp platform 3 compille error
```
C:\Users\j.jirutka\Documents\Arduino\libraries\WireGuard-ESP32\src\wireguardif.c:51:10: fatal error: tcpip_adapter.h: No such file or directory
   51 | #include "tcpip_adapter.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

alfersi, 2024-07-20:
I had the exact same issue, the only way to get it to compile is switching to a platform version <3.0.0, the reason being that version 3.0.0 and higher doesn't include the TCP/IP Adapter library anymore, having been deprecated in favour of ESP NETIF. I understand that for this library to work with ESP NETIF it  would require a complete review a rewrite following the migration guide, however I don't know if there's a fork that already includes these changes or if anyone is willing to do it. My suggestion would be to use platformIO where you can specifify the platform version, and hence the ESP-IDF version, in this case Arduino ESP32 version 2.0.17 uses ESP-IDF v4.4.7, in platformIO you can specify:

`[env:my-esp32]
platform = espressif32 @ 5.3.0`

which uses ESP-IDF v4.4.3 and should compile nicely

nmingotti, 2024-08-10:
a pity, if anybody is willing to port the software to make it compile and work again in the Arduino IDE (Linux) for Heltec Wifi Kit 32(V3)  (which i use most recently) will be ready to give and hand financially say 100€.

martinusGH, 2025-01-03:
I had the same issue when migrating from Arduino 2.x to 3.0.7. I managed to patch the source code myself and it worked again until I upgraded to Arduino 3.1.0. Broken again, again several hours of debugging, but managed to fix it again. Needs a lot of Mutexes wrapped around many calls to net_if.
Unfortunately this library seems to be abandoned as it has not seen any updates for three years.

alpintrekker, 2025-01-09:
@martinusGH would you mind to share your results, maybe in a forked repository?

martinusGH, 2025-01-09:
I've trashed my github development setup a few years ago, and not really planning to restart that. I can try to add the two patched files in here, hope you are able to use those into a working solution. My skills seemed sufficient to make this library work again, but really someone with deep knowledge on ESP32-IDF based networking should try to take over the maintenance of this library.
[WireGuard.zip](https://github.com/user-attachments/files/18358715/WireGuard.zip)

martinusGH, 2025-01-11:
I cannot remember changing that file, but let me just upload the entire library that I have now
[WireGuard-ESP32.zip](https://github.com/user-attachments/files/18385287/WireGuard-ESP32.zip)
I double checked with the provided sample from the library "disconnect", compiles without issues on my end.

pierrejay, 2025-04-05:
If useful to anyone, I saw that ESPHome had done an implementation that works on the 3.0+ core, so I made a port that looks to be working fine with better performance than the patched library (typically getting ~30ms better round-trip time on incoming pings & HTTP requests).

You can find it [here](https://github.com/pierrejay/esp32-tailbridge/tree/main/esp32-arduino-example) within another project I have done using the lib.

N.B. : some build flags must be added to `platformio.ini` (same as in the ESPHome implementation):
```ini
build_flags = 
  -DCONFIG_WIREGUARD_MAX_SRC_IPS=4
  -DCONFIG_WIREGUARD_MAX_PEERS=1
```

freitas666, 2025-04-06:
Hello guys,

This Wireguard-ESP32.zip works in Esp32-s3? I am trying compile and have problems with tcpip_adapter.h..

Forgive me, i am not a pro on this...

pierrejay, 2025-04-06:
Yes, tested the archive with both c3 and s3 and it was working fine. `tcpip_adapter.h` is no longer included in this version so you might be using the wrong version of the lib.

pierrejay, 2025-04-06:
Personally I am using PlatformIO (`pioarduino` platform) not Arduino IDE. 

I believe you should first uninstall the WireGuard-ESP32 library in Arduino IDE's Library Manager and after only, add the .zip in `Sketch > Include Library`. 

I just tried it with martinusGH's version and got some compilation errors. One function definition was missing in `wireguard.h`. Try this archive, it works for me with `XIAO_ESP32S3` with core 3.2.0. I can see the ESP32 connecting to my server and it responds to pings.

[WireGuard-ESP32-bis.zip](https://github.com/user-attachments/files/19623865/WireGuard-ESP32-bis.zip)

freitas666, 2025-04-07:
> Personally I am using PlatformIO (`pioarduino` platform) not Arduino IDE.
>
> I believe you should first uninstall the WireGuard-ESP32 library in Arduino IDE's Library Manager and after only, add the [...]

Thank you, i will try it and soon back with results :D

freitas666, 2025-04-07:
Hello @pierrejay i followed this steps that you send me and my code works perfectly! no more errors, compile good and works connecting on my wireguard server!

**Thank you so much!**